cnlohr/lolra #2: Request: Publish power consumption data if available.

happytm, 2024-03-26:
Thank you for your groundbreaking efforts to transmit data without LoRa hardware.

It will be useful to know power consumption of transmitting device. I expect it to be ultra low power compare to LoRa radio hardware. 

If you have power consumption data or guess about power consumption please let us know.

Thanks again.

zuyan9, 2024-03-30:
It won't be more power efficient than a LoRa radio chip because most of the power transmitted is in non-LoRa frequencies.

unicab369, 2024-04-07:
[image]
[image]

This is the result tested on the ESP-01. Had some issue with the ESP12. I may look into that later but I don't see the result could have been any different. some of the power consumption could have come from the LED as well during transmission.
for reference ESP-Now uses on average about 57mA during transmission.

happytm, 2024-04-07:
@unicab369 Thank you for posting power profile.

happytm, 2024-04-19:
@unicab I have few question:

- How many bytes  were you sending for above test?
- Can you publish compiled file to upload to ESP for testing purpose?
- Am I reeading it right that it takes only 1mA extra to transmit? If so, it is very good result in terms of power consumption.
- Would the range be better than ESPNow?
Thanks.

happytm, 2024-04-23:
@cnlohr Thank you for your response.

happytm, 2024-04-23:
From your range tests it is 2580 meters of maximum range you got with ESP32-S2 which is impressive. The first step to test this is to compile the code and upload it to ESP32 but many people like me have problem crossing this hurdle. I have expertise in antenna  and filter design stuff and I wanted to increase the range without power consumption penalty.

Can you give us step by step instruction for compiling and uploading this code? If not, can you publish compiled code file so we can use for test purpose with different antenna design? I can make up to 27 dBi gain directional antenna or 15 dBi omni directional antenna easily. I wanted to test these antenna with your code to test the limits.

Thanks.

cnlohr, 2024-04-24:
Indeed it is quite the hurdle - You will need to follow the instructions here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/get-started/index.html

Then, use the terminal window that comes up after installation where you can compile projects, and navigate to the folder in this tree that contains the CMakeLists.txt. then from there you can say

```
idf.py flash
```

And it will work (ok, realistically it won't work because the IDF environment is so fragile.  But, it is work-through-able)

happytm, 2024-05-12:
@cnlohr  Would you please simply publish binary file ? 

Thanks.