cnlohr/lolra #13: Compiler Error on Windows for Ch32v203

anlpereira, 2024-11-28:
Hi, I'm a windows user and I'm trying to compile using GCC 6.3.0 but there are some compile-time errors. I tryied to fix them but without success. I'm really want to implement this project here. I got very curious about it. Thank you very much.

rf_data_gen.c:68:34: error: initializer element is not constant
  const double center_frequency = 904.5 + tune_mhz;
                                  ^~~~~
In file included from rf_data_gen.c:86:0:
../../lib/rf_data_gen.h:106:28: error: initializer element is not constant
 const double chirp_begin = center_frequency-bw/2;
                            ^~~~~~~~~~~~~~~~
../../lib/rf_data_gen.h:107:26: error: initializer element is not constant
 const double chirp_end = center_frequency+bw/2;
                          ^~~~~~~~~~~~~~~~
../../lib/rf_data_gen.h:118:28: error: initializer element is not constant
 const double sampletotal = ( chirp_length_seconds * sample_rate * 1000000 );

cnlohr, 2024-11-28:
You must use at least GCC 10 or newer. https://gnutoolchains.com/risc-v/

anlpereira, 2024-11-28:
After I install https://gnutoolchains.com/risc-v/ and add bin to PATH.

C:\Users\anlpe\lolra\ch32v\ch32v203-lora>make
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V203 -DMCU_PACKAGE=2 -DTARGET_MCU_LD=2 -DTARGET_MCU_MEMORY_SPLIT= C:/Users/anlpe/ch32v003fun/ch32v003fun/ch32v003fun.ld > C:/Users/anlpe/ch32v003fun/ch32v003fun/generated_CH32V203F6P6_.ld
gcc -o rf_data_gen rf_data_gen.c -lm
process_begin: CreateProcess(NULL, gcc -o rf_data_gen rf_data_gen.c -lm, ...) failed.
make (e=2): O sistema não pode encontrar o arquivo especificado.
make: *** [rf_data_gen] Error 2

anlpereira, 2024-11-29:
I got it. I compiled the file. Now I need to generate the .hex file to flash with the WCHISPStudio tool. Who can help me? Thanks a lot

DonSMectron, 2024-11-29:
I use VS Code and PlatformIO. 

Using "ch32v\ch32v003fun\blink" as an example, I made these small changes.

Copy the folder ch32v203-goertzel to 
"lolra-master\ch32v\ch32v003fun\examples_v20x" 

Add the project to PlatformIO.ini
lolra-master\ch32v\ch32v003fun\PlatformIO.ini

[env:v20x_ch32v203-goertzel]
extends = fun_base_203
build_src_filter = ${fun_base.build_src_filter} +<examples_v20x/ch32v203-goertzel>

Inside "adcgoertzel.c"
Change:
#include "./usb_config.h"
#include "../ch32v003fun/examples_v20x/otg_device/otgusb.h"
To:
#include "usb_config.h"
#include "../otg_device/otgusb.h"

The Calculator.html file works in chrome. I was able to hear the radio station 1050AM ( near Ann Arbor MI ).

I also got the 128 x 128 OLED display working with it!

For Windows use the driver from "wch-ic.com"

anlpereira, 2024-11-29:
I don’t have plataformIO. I can flash the ch32 using my board, but I don’t know how to generate the hex files. I have this black board on the left that is the ch32 and these others two I did in my CNC to receive the signals. One uses an attiny85 and the other an atmega328, both with lora modules. 
[image]

DonSMectron, 2024-11-29:
Hmm, yes I see. Well it looks like platformIO generates firmware.elf type files that are then flashed to the chip. Different technology. How to flash the chip now? Have you looked at the Arduino IDE?
https://www.hackster.io/patrick-fitzgerald2/program-ch32v003-risc-v-with-arduino-ide-135f6f

anlpereira, 2024-11-30:
I got it solved. Thank you very much for any effort. The solution was to add a line inside MakeFile after, 'TARGET_MCU:=CH32V203'
that is 'TARGET_MCU_PACKAGE:=C8'

closed 2024-11-30 by anlpereira